ucla-mobility/OpenCDA #105: Is reinforcement learning supported?

xiaoyucheri, 2021-07-30:
Thanks for creating this great project! I saw OpenCDA has some rule-based algorithms as planning module. So does it also support reinforcement learning behavior planning? Thanks!

DerrickXuNu, 2021-07-30:
Yes, OpenCDA supports RL development. Pass the RL states to the behavior agent through function `update_information()`, execute the action in `run_step()` and collect the rewards in `plan_debuger()` or `EvaluationManager()`. If you are using deep RL, you can put your model in `MLManager` and add it as an attribute in your behavior agent class. Check https://opencda-documentation.readthedocs.io/en/latest/md_files/customization.html to see more details.

xiaoyucheri, 2021-07-31:
That works, thanks!

closed 2021-07-31 by xiaoyucheri