ucla-mobility/OpenCDA #117: RuntimeError: time-out of 10000ms while waiting for the simulator

luckynote, 2021-08-02:
` python opencda.py -t platoon_joining_2lanefree_cosim`
OpenCDA Version: 0.1.0
load opendrive map '2lane_freeway_simplified.xodr'.
Traceback (most recent call last):
  File "/home/idriver/liutao/github/OpenCDA/opencda/scenario_testing/platoon_joining_2lanefree_cosim.py", line 42, in run_scenario
    sumo_file_parent_path=sumo_cfg)
  File "/home/idriver/liutao/github/OpenCDA/opencda/scenario_testing/utils/cosim_api.py", line 64, in __init__
    cav_world)
  File "/home/idriver/liutao/github/OpenCDA/opencda/scenario_testing/utils/sim_api.py", line 114, in __init__
    self.world = load_customized_world(xodr_path, self.client)
  File "/home/idriver/liutao/github/OpenCDA/opencda/scenario_testing/utils/customized_map_api.py", line 54, in load_customized_world
    enable_mesh_visibility=True))
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "opencda.py", line 56, in <module>
    main()
  File "opencda.py", line 51, in main
    scenario_runner(opt, config_yaml)
  File "/home/idriver/liutao/github/OpenCDA/opencda/scenario_testing/platoon_joining_2lanefree_cosim.py", line 86, in run_scenario
    eval_manager.evaluate()
UnboundLocalError: local variable 'eval_manager' referenced before assignment

luckynote, 2021-08-02:
And other command has same problem.

DerrickXuNu, 2021-08-02:
what is the carla version you are using?

DerrickXuNu, 2021-08-02:
> And other command has same problem.

Which command do you refer? Could you be more specific?

luckynote, 2021-08-02:
> what is the carla version you are using?

[Ubuntu] CARLA_0.9.11.tar.gz
[Ubuntu] AdditionalMaps_0.9.11.tar.gz

luckynote, 2021-08-02:
> > And other command has same problem.
> 
> Which command do you refer? Could you be more specific?
Quick Start: Single Vehicle Test & Cooperative Driving Test.

DerrickXuNu, 2021-08-02:
It seems that you can not connect to the CARLA server. Can you make sure:
1) Your CARLA server is running normally
2) You don't have any other scripts using the same Client port running
If both fits, can you try to restart the CARLA server?

closed 2021-08-03 by luckynote

wx983, 2022-12-25:
> 

But how to see the present version of carla?   I used 'git checkout tags/0.9.12 -b <branch_name>'  when installing it, but I don't know how to check the version