ucla-mobility/OpenCDA #120: RuntimeError: opendrive could not be correctly parsed

yanghao, 2021-08-08:
Not sure if I missed anything but I cannot get the basic example working.

Command for starting carla server:
```
/opt/carla-simulator/CarlaUE4.sh 
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
```
command for starting opencda:
```
$ python opencda.py -t single_2lanefree_carla
OpenCDA Version: 0.1.0
load opendrive map '2lane_freeway_simplified.xodr'.
Traceback (most recent call last):
  File "/home/yanghao/external/OpenCDA/opencda/scenario_testing/single_2lanefree_carla.py", line 35, in run_scenario
    cav_world=cav_world)
  File "/home/yanghao/external/OpenCDA/opencda/scenario_testing/utils/sim_api.py", line 114, in __init__
    self.world = load_customized_world(xodr_path, self.client)
  File "/home/yanghao/external/OpenCDA/opencda/scenario_testing/utils/customized_map_api.py", line 54, in load_customized_world
    enable_mesh_visibility=True))
RuntimeError: opendrive could not be correctly parsed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "opencda.py", line 56, in <module>
    main()
  File "opencda.py", line 51, in main
    scenario_runner(opt, config_yaml)
  File "/home/yanghao/external/OpenCDA/opencda/scenario_testing/single_2lanefree_carla.py", line 75, in run_scenario
    eval_manager.evaluate()
UnboundLocalError: local variable 'eval_manager' referenced before assignment
```

DerrickXuNu, 2021-08-08:
Thanks for using OpenCDA! To help with your questions, could you provide these information?
1) What is your Carla simulator version?
2) What is the Carla API version installed in your opencda conda env?
3) What is your OS?

yanghao, 2021-08-13:
> Thanks for using OpenCDA! To help with your questions, could you provide these information?
> 
> 1. What is your Carla simulator version?
See above post, Carla prints its version during startup: 4.24.3-0+++UE4+Release-4.24 518 0

> 2. What is the Carla API version installed in your opencda conda env?
I don't know how to check this? any command I can run from the conda env?

> 3. What is your OS?
See above post, ubuntu 2004

yanghao, 2021-08-13:
> > Thanks for using OpenCDA! To help with your questions, could you provide these information?
> > 
> > 1. What is your Carla simulator version?
> >    See above post, Carla prints its version during startup: 4.24.3-0+++UE4+Release-4.24 518 0
The carla-simulator package version from apt show: 0.9.11.

> > 1. What is the Carla API version installed in your opencda conda env?
> >    I don't know how to check this? any command I can run from the conda env?
If you are referring to the python carla library version in conda: 0.9.11

DerrickXuNu, 2021-08-13:
Thanks for your information! I didn't test on Carla installed by apt before and not sure whether that is the reason. Could you try to download the release package version https://github.com/carla-simulator/carla/releases/tag/0.9.11 , set your CARLA_HOME to this downloaded package  and then reinstall opencda again?

yanghao, 2021-08-13:
Still the same result same error message.

Just to be sure, the steps I take to run the test (after the installation of course):

`$ ./CarlaUE4.sh # run carla
`
Then from a different console:
```
$ cd OpenCDA
$ conda activate opencda
$ python opencda.py -t single_2lanefree_carla
```

Hopefully, these are the correct steps.

DerrickXuNu, 2021-08-13:
This is a wired issue... yeah your steps are correct. Could you try to run python opencda.py -t single_town06_carla --apply_ml to see whether it works?

yanghao, 2021-08-14:
with `--apply_ml`, the "single_town06_carla" example works.
[image]

without `--apply_ml`, it gives a different error message:

```
$ python opencda.py -t single_town06_carla 
OpenCDA Version: 0.1.0
Creating single CAVs.
Traceback (most recent call last):
  File "/home/yanghao/external/OpenCDA/opencda/scenario_testing/single_town06_carla.py", line 32, in run_scenario
    scenario_manager.create_vehicle_manager(application=['single'])
  File "/home/yanghao/external/OpenCDA/opencda/scenario_testing/utils/sim_api.py", line 203, in create_vehicle_manager
    data_dumping=data_dump)
  File "/home/yanghao/external/OpenCDA/opencda/core/common/vehicle_manager.py", line 103, in __init__
    data_dumping)
  File "/home/yanghao/external/OpenCDA/opencda/core/sensing/perception/perception_manager.py", line 323, in __init__
    'If you activate the perception module, '
SystemExit: If you activate the perception module, then apply_ml must be set to true inthe argument parser to load the detection DL model.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "opencda.py", line 56, in <module>
    main()
  File "opencda.py", line 51, in main
    scenario_runner(opt, config_yaml)
  File "/home/yanghao/external/OpenCDA/opencda/scenario_testing/single_town06_carla.py", line 63, in run_scenario
    eval_manager.evaluate()
UnboundLocalError: local variable 'eval_manager' referenced before assignment
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 495 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 496 (sensor.other.imu) 
terminate called without an active exception
Aborted (core dumped)
```

DerrickXuNu, 2021-08-14:
Thanks for the try. 

First, running town06 without `apply_ml` flag showing error is **expected** . This is because in the `single_town06_carla.yaml`, the activation flag of perception module is set to true by default, which requires pytorch library.  By setting `--apply_ml` flag, OpenCDA will load pytorch library. We did this design so that the users who don't focus on perception won't need to install pytorch library.

Second, since you can run town06 example successfully, **it means the opencda is installed correctly and running successfully.** There is something operating abnormally  with the opendrive parser in your Carla simulator, which is hard to debug. Some of the Carla users have met the similar issues as yours: https://github.com/carla-simulator/carla/issues/3269 and the Carla team hasn't came up with a solid solution with this issue.  Since OpenCDA itself is functioning well in your computer, I'd suggest running the simulations on the default maps(such as town06 and town05) instead of the provided customized map. OpenCDA documentation has a very detailed description of how to construct your own scenarios: https://opencda-documentation.readthedocs.io/en/latest/md_files/yaml_define.html

closed 2021-08-17 by DerrickXuNu

yanghao, 2021-08-17:
@DerrickXuNu Thanks a lot.

tuntun-sd, 2021-08-25:
If you experience the error opendrive could not be correctly parsed, ensure that there are write permissions on the CarlaUE4/Content/Carla/Maps/OpenDrive/ directory. This is required by the server to parse the xodr file correctly.[https://carla.readthedocs.io/en/latest/adv_opendrive/](url)

yanghao, 2021-08-29:
@tuntun-sd Indeed this is the problem. carla refuse to run with root permission on GNU/Linux, so this particular carla folder has to be made writable by all users (which is not very nice ...). Thanks for the information.

sjtulhm, 2023-03-08:
> @DerrickXuNu多谢。
Town06 is not found in your CARLA repo! Please download all town maps to your CARLA repo!，
我想请问一下这个CARLA repo是在哪儿呀，我需要把那个地图文件解压到哪儿，上面是我的报错内容。