ucla-mobility/OpenCDA #121: Sumo encountered the following

XIAOXIAOXIXIA, 2021-08-12:
I ran the third example of quick start on the opencda official website, and Sumo encountered the following error：simulation ended at time ：0.05.

DerrickXuNu, 2021-08-12:
Hi, could you provide more information of the error on the Sumo panel/ console output? Also what is your sumo and Carla version?

XIAOXIAOXIXIA, 2021-08-12:
> Hi, could you provide more information of the error on the Sumo panel/ console output? Also what is your sumo and Carla version?

Sumo is 1.2.0 and Carla is 0.9.11.
CMD'information:C:\Users\86188\Desktop\opencda\openCDA\OpenCDA-main\OpenCDA-main>python opencda.py -t platoon_joining_2lanefree_cosim
OpenCDA Version: 0.1.0
<module 'opencda.scenario_testing.platoon_joining_2lanefree_cosim' from 'C:\\Users\\86188\\Desktop\\opencda\\openCDA\\OpenCDA-main\\OpenCDA-main\\opencda\\scenario_testing\\platoon_joining_2lanefree_cosim.py'>
load opendrive map '2lane_freeway_simplified.xodr'.
Creating platoons/
Creating single CAVs.
C:\Users\86188\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\fromnumeric.py:3441: RuntimeWarning: Mean of empty slice.
  out=out, **kwargs)
C:\Users\86188\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\_methods.py:189: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
C:\Users\86188\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\fromnumeric.py:3441: RuntimeWarning: Mean of empty slice.
  out=out, **kwargs)
C:\Users\86188\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\_methods.py:189: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
C:\Users\86188\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\fromnumeric.py:3441: RuntimeWarning: Mean of empty slice.
  out=out, **kwargs)
C:\Users\86188\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\_methods.py:189: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
C:\Users\86188\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\fromnumeric.py:3441: RuntimeWarning: Mean of empty slice.
  out=out, **kwargs)
C:\Users\86188\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\_methods.py:189: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
Localization Evaluation Done.
Kinematics Evaluation Done.
Platooning Evaluation Done.
destroying carla actor
destroying sumo actor
SUMO's information:Simulation ended at time:0.05.Reason:traci requested termination.

DerrickXuNu, 2021-08-12:
Your sumo version maybe too old， try to upgrade to 1.9.0 and see whether it works.

XIAOXIAOXIXIA, 2021-08-15:
> Your sumo version maybe too old， try to upgrade to 1.9.0 and see whether it works.

I updated sumo to 1.9.2,I ran the third example of quick start on the opencda official website, and Sumo encountered the following error：simulation ended at time ：12.65.
CMD'information:C:\Users\86188\Desktop\opencda\openCDA\OpenCDA-main\OpenCDA-main>python opencda.py -t platoon_joining_2lanefree_cosim
OpenCDA Version: 0.1.0
<module 'opencda.scenario_testing.platoon_joining_2lanefree_cosim' from 'C:\\Users\\86188\\Desktop\\opencda\\openCDA\\OpenCDA-main\\OpenCDA-main\\opencda\\scenario_testing\\platoon_joining_2lanefree_cosim.py'>
load opendrive map '2lane_freeway_simplified.xodr'.
Creating platoons/
Creating single CAVs.
WARNING:root:sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.lincoln.mkz2017
WARNING:root:sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.lincoln.mkz2017
WARNING:root:sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.jeep.wrangler_rubicon
WARNING:root:sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.mustang.mustang
WARNING:root:sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.nissan.patrol
WARNING:root:sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.mini.cooperst
WARNING:root:sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.mini.cooperst
WARNING:root:sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.seat.leon
Destination Reset!
Destination Reset!
Destination Reset!
delete waypoint!
Destination Reset!
Destination Reset!
Destination Reset!
Destination Reset!
delete waypoint!
Destination Reset!
delete waypoint!
Simulation is Over
Localization Evaluation Done.
Kinematics Evaluation Done.
Platooning Evaluation Done.
destroying carla actor
destroying sumo actor
SUMO's information:Simulation ended at time:12.65.Reason:traci requested termination.

DerrickXuNu, 2021-08-15:
That's not an error, it is just the simulation is over as the AV reaches the target location. Check the console carefully and you will see a sentence "The simulation is over!", which means this simulation successfully finishes the task and clean out the spawn vehicles. Your issue has been fixed.

closed 2021-08-15 by DerrickXuNu

XIAOXIAOXIXIA, 2021-08-15:
我参考的是官网的启动文件，这一个例子他想表明的是匝道车辆和主道车辆完成汇流，但是仿真在12.65秒的时候就终止了，并且在仿真过程中没有观察到匝道的车辆只观测到了主道的车辆，这一个和官网的例子不太匹配。 The example I referred to was the startup file of the official website. It wanted to show that the vehicles on the ramp and on the main road would merge, but the simulation stopped at 12.65 seconds and I only observed  vehicles on the main road no vehicles on the ramp road, which did not match the one on the official website.

DerrickXuNu, 2021-08-15:
可以把你的运行指令发一下吗？如果可以录下屏看下video什么样子发给我。rxx3386@ucla.edu