ucla-mobility/OpenCDA #122: ERROR: single_2lanefree_carla.py not found under opencda/scenario_testing

JackRen88, 2021-08-16:
Hello.
When I run the code `python opencda.py -t single_2lanefree_carla`.there is a error below:
`ERROR: single_2lanefree_carla.py not found under opencda/scenario_testing`. How should I solved it?

DerrickXuNu, 2021-08-16:
It seems you may not tightly follow the installation instructions. Can you make sure when you install OpenCDA, you ran these three command?

conda env create -f environment.yml
conda activate opencda
python setup.py develop

Also, before you run any script, please active the opencda environment first.

DerrickXuNu, 2021-08-21:
I would regard the issue is solved if no response after 3 more days. Please let me know if you still have issue.

closed 2021-08-23 by DerrickXuNu