ucla-mobility/OpenCDA #125: UnboundLocalError: local variable 'eval_manager' referenced before assignment

1790387702, 2021-08-24:
Hello, everyone. Have you ever encountered this problem? Please help me, Thanks!!

tuntun-sd, 2021-08-24:
I met the same problem. And i also met another problem:

> RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000

DerrickXuNu, 2021-08-24:
When you saw this issue, the root cause is not exactly this issue. Typically, it is caused by some other problems that happened before the evaluation manager is created, and when the codes go to the finally block this issue will raise as evaluation manager hasn't been created yet. So please paste the full error message here so I can help you.

DerrickXuNu, 2021-08-24:
> I met the same problem. And i also met another problem:
> 
> 
> 
> > RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
> 
> 

Please refer to this closed issue.

https://github.com/ucla-mobility/OpenCDA/issues/117

1790387702, 2021-08-24:
Because I have not install carla in canda, and then I did it ,but now the question is changed as below..
OpenCDA Version: 0.1.0
load opendrive map '2lane_freeway_simplified.xodr'.
Creating single CAVs.
Traceback (most recent call last):
  File "/home/fxd/Project/OpenCDA/opencda/scenario_testing/single_2lanefree_carla.py", line 43, in run_scenario
    map_helper=map_api.
  File "/home/fxd/Project/OpenCDA/opencda/scenario_testing/utils/sim_api.py", line 212, in create_vehicle_manager
    vehicle_manager.update_info()
  File "/home/fxd/Project/OpenCDA/opencda/core/common/vehicle_manager.py", line 174, in update_info
    objects = self.perception_manager.detect(ego_pos)
  File "/home/fxd/Project/OpenCDA/opencda/core/sensing/perception/perception_manager.py", line 402, in detect
    objects = self.deactivate_mode(objects)
  File "/home/fxd/Project/OpenCDA/opencda/core/sensing/perception/perception_manager.py", line 563, in deactivate_mode
    objects)
  File "/home/fxd/Project/OpenCDA/opencda/core/sensing/perception/o3d_lidar_libs.py", line 138, in o3d_visualizer_show
    vis.update_geometry(point_cloud)
TypeError: update_geometry(): incompatible function arguments. The following argument types are supported:
    1. (self: open3d.open3d.visualization.Visualizer) -> bool

Invoked with: Visualizer with name 87, geometry::PointCloud with 2133 points.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "opencda.py", line 56, in <module>
    main()
  File "opencda.py", line 51, in main
    scenario_runner(opt, config_yaml)
  File "/home/fxd/Project/OpenCDA/opencda/scenario_testing/single_2lanefree_carla.py", line 75, in run_scenario
    eval_manager.evaluate()
UnboundLocalError: local variable 'eval_manager' referenced before assignment
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 88 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 89 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 90 (sensor.camera.rgb) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 91 (sensor.lidar.ray_cast)

DerrickXuNu, 2021-08-24:
Can you show me your running command,  OS system version and Carla version?

DerrickXuNu, 2021-08-24:
Also, did you install opencda with the yaml file we provided?

1790387702, 2021-08-25:
@DerrickXuNu Thanks a lot. It seems I have find the root cause.Because in the environment.yaml and requirements.txt, the open3d has no version information, so pip default install 0.13.0. when I install open3d not higher than 0.10.x, The Porblem has solved.So Maybe there is some incompatibility problem..

DerrickXuNu, 2021-08-25:
I am using 0.13.0 and it works fine. I guess there is incompatibility between your system and Open3d. Anyway, please let me know if you have further questions

closed 2021-08-25 by DerrickXuNu

tuntun-sd, 2021-08-25:
> > I met the same problem. And i also met another problem:
> > > RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
> 
> Please refer to this closed issue.
> 
> #117

Thanks a lot!

DerrickXuNu, 2023-03-08:
Did you start Carla simulators first?

sjtulhm, 2023-03-09:
> Did you start Carla simulators first?
我运行：
‘/CarlaUE4.sh' 
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps，能正常启动的。
我运行您上述的python opencda.py-t single_town06_carla --apply_ml，会出现Town06 is not found in your CARLA repo! Please download all town maps to your CARLA repo! 这样的报错，我已经下载了那个additional map也替换到了原来的文件夹里。

sjtulhm, 2023-03-13:
> 
Dr. Xu, could you take the time to answer my question

sjtulhm, 2023-03-13:
> Did you start Carla simulators first?
Dr. Xu, could you take the time to answer my question

wissage, 2025-05-08:
> > Did you start Carla simulators first?
> > Dr. Xu, could you take the time to answer my question

Hello, i met the same problem. 
Town05 is not found in your CARLA repo! Please download all town maps to your CARLA repo! 
Have you solved the problem?
Thanks a lot!

wissage, 2025-05-08:
> Did you start Carla simulators first?
> 我运行：
> ‘/CarlaUE4.sh'
> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps，能正常启动的。
> 我运行您上述的python opencda.py-t single_town06_carla --apply_ml，会出现Town06 is not [...]

Hello, i met the same problem.
Town05 is not found in your CARLA repo! Please download all town maps to your CARLA repo!
Have you solved the problem?
Thanks a lot!

wjthkx, 2025-09-03:
> Did you start Carla simulators first?
> 我运行：
> ‘/CarlaUE4.sh'
> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps，能正常启动的。
> 我运行您上述的python opencda.py-t single_town06_carla --apply_ml，会出现Town06 is not [...]

I have encountered this problem before. 
Did you import maps using the script? 
After using ImportAssets.sh to import maps in Carla, it was solved. 
I hope this helps you  : )