ucla-mobility/OpenCDA #127: Ubuntu16.04 can NOT run Two-lane highway test

CharlesWolff6, 2021-09-14:
Hi, 
   Thanks for the great work

   I try to run the single_2lanefree_carla on Ubuntu 16.04, but it failed:

----------------------------------------------------------------------------
~/OpenCDA$ python opencda.py -t single_2lanefree_carla
OpenCDA Version: 0.1.0
Traceback (most recent call last):
  File "opencda.py", line 56, in <module>
    main()
  File "opencda.py", line 40, in main
    testing_scenario = importlib.import_module("opencda.scenario_testing.%s" % opt.test_scenario)
 ...
    import open3d as o3d
  File "/home/anaconda3/envs/opencda/lib/python3.7/site-packages/open3d/__init__.py", line 56, in <module>
    _CDLL(str(next((_Path(__file__).parent / 'cpu').glob('pybind*'))))
  File "/home/anaconda3/envs/opencda/lib/python3.7/ctypes/__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by /home/anaconda3/envs/opencda/lib/python3.7/site-packages/open3d/cpu/pybind.cpython-37m-x86_64-linux-gnu.so)
----------------------------------------------------------------------------

I searched in google and found that it maybe the problem of open3d  which uses glibc 2.27
Ubuntu16.04 seems to not be supported anymore. Ubuntu 16.04 use only glibc 2.23

https://github.com/isl-org/Open3D/issues/1898

DerrickXuNu, 2021-09-14:
Hi, thanks for reporting this. Actually, I think you can just downgrade your open3d to a previous version to give a try first.

CharlesWolff6, 2021-09-16:
Hi, 
    I downgrade the open3d to 0.9.0.0 
    It has another error:

  File "/home/OpenCDA/opencda/core/sensing/perception/perception_manager.py", line 345, in __init__
    self.o3d_vis = o3d_visualizer_init(vehicle.id)
  File "/home/OpenCDA/opencda/core/sensing/perception/o3d_lidar_libs.py", line 104, in o3d_visualizer_init
    top=270)
RuntimeError: [Open3D ERROR] GLFW Error: GLX: Failed to create context: GLXBadFBConfig

    I think maybe the open3d version is too old. So I modified the "single_2lanefree_carla.yaml" line 20,  change lidar_visualize from true to false.   It worked.

DerrickXuNu, 2021-09-16:
What gpu and driver you are using?

closed 2021-09-18 by DerrickXuNu

6Lackiu, 2022-04-26:
Hello, my remote server system is ubuntu16.04, and I also encountered this problem with GLIBC_2.27.
After reading your previous answer, I lowered the open3d version from 0.13 to 0.9.0, but this error occurred when running`platoon_joining_2lanefree_carla.py` , how can I solve it?
Carla: 0.9.11
GPU: GeForce RTX2080
Driver Version: 418.43
CUDA Version: 10.1
[image]

6Lackiu, 2022-04-26:
> Hello, my remote server system is ubuntu16.04, and I also encountered this problem with GLIBC_2.27. After reading your previous answer, I lowered the open3d version from 0.13 to 0.9.0, but this error occurred when running`platoon_joining_2lanefree_carla.py` , how can I solve it? Looking forward to your reply! Carla: 0.9.11 GPU: GeForce RTX2080 Driver Version: 418.43 CUDA Version: 10.1 [image]

I delete ‘.’ in `from .open3d import * # py2 py3 compatible`
This problem no longer reports an error, but prompts
`ERROR: platoon_joining_2lanefree_carla.py not found under opencda/scenario_testing`