ucla-mobility/OpenCDA #139: Running opencda in docker support

jewes, 2021-11-22:
This is not a real issue, but just some notes for those who want to running opencda in docker environment.

1. Base Docker Image: I already have a base docker image(ubuntu 18.04) with carla client lib(0.9.11) installed. ie. `import carla` will not generate any error messages. 
2. OpenCDA installation: Get a copy of the source code, and mount it to the docker container based on image in the previous step using the docker `-v` options. So you'll get access to the opencda source in the docker container.
3. X11 support: using docker run option `-v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY`

Possible errors:
1. In the container shell, try to run a scenario, for example. the single_2lanefree_carla, you may get some libs (limSM.so, libGL.so) missing messages. To fix those errors: using `apt update && apt install -y libsm6 libgl1-mesa-glx` to install the dependencies.
2. You may get some error like "X error: BadShmSeg, blabla", set environment "export QT_X11_NO_MITSHM=1" in the container.

If you see some other errors, leave a message here, I'll see if I can help.

DerrickXuNu, 2021-11-22:
Thanks a lot for addressing this!

wdf19961118, 2022-03-27:
Could you release your base docker image? Thanks@jewes

DerrickXuNu, 2022-03-27:
@jewes  Hi Jewes, welcome to contribute a PR for the docker image~