ucla-mobility/OpenCDA #142: The errors about 'torch.cuda' and  'eval_manager'

aslirey, 2021-12-17:
Hello,

It is really great work! I am interested in co-simulation with sumo. While running it, I have encountered with errors. Could you please help me? 

Kind regards,
[image]

aslirey, 2021-12-17:
Can I get RGB sensor data during the simulation to use them for reinforcement learning later? Thanks...

DerrickXuNu, 2021-12-18:
Thanks for being interested in OpenCDA! 
1) Answers to your error:  What's your pytorch version? Pytorch >= 1.8 is required
2) Yes, OpenCDA support streaming rgb sensor data. Check the document: https://opencda-documentation.readthedocs.io/en/latest/md_files/developer_tutorial.html#perceptionmanager and the code: https://github.com/ucla-mobility/OpenCDA/blob/main/opencda/core/sensing/perception/perception_manager.py#L29

aslirey, 2021-12-18:
Thank you very much. I have upgraded pytorch to 1.10.0. Now it is running.

closed 2021-12-18 by aslirey