ucla-mobility/OpenCDA #154: License clarification questions. Can we distribute generated data?

lynnpepin, 2022-03-25:
Hello! OpenCDA is a really good project and I am enjoying using it. I appreciate the thorough documentation and the clear code style.

I understand the license permits researchers to use the code with credit. But we cannot make our forked code public, and if we do, the license is terminated and we can no longer use OpenCDA. (I.e. A normal "non-distrib" license as I understand it.)

I wanted to clarify the following:

1. Can we publicly share **scenario definitions?** E.g. A simulation yaml and the corresponding Python code.
2. Does this apply to pull requests? (I think this license would technically prohibit others from making pull requests. I have nothing ready to contribute, but I just want to be sure.)
3. Can we publicly share **generated data?** E.g. Just the dataset output from OpenCDA, and none of the code.

majiaqi, 2022-03-25:
Hi
Thanks for your interest. You are right about the license, generally, except item 2. We actually encourage you to submit your PR and we welcome you to join and contribute to the OpenCDA repository. In fact, multiple PRs from outside the team have been approved and merged and we expect many more coming soon. Regarding item 3, since the data is technically a derivative of the OpenCDA platform, so it may be also restrictive if you use OpenCDA directly to generate the datasets. We may need to know a bit more about your scenario to determine that. However, one way to bypass the license issue here is to work with us to generate additional datasets and contribute back to OpenCDA. We can figure out flexible ways for you to open the datasets. As you see, our key goal is to have researchers who use OpenCDA to join the community and contribute back.

lynnpepin, 2022-03-28:
Understood, thank you for clarifying! I'm happy to have this clarified, and will definitely be submitting PRs if I have useful contributions! :)

closed 2022-03-28 by lynnpepin