ucla-mobility/OpenCDA #155: not found under opencda/scenario_testing

williamhatch, 2022-04-18:
python opencda.py -tsingle_town05_cosim -v 0.9.11 and got ERROR: single_town05_cosim.py not found under opencda/scenario_testing, any idea? @DerrickXuNu

closed 2022-04-18 by williamhatch

DerrickXuNu, 2022-04-18:
Hi, I saw you closed the issue. Have you solved the issue?

6Lackiu, 2022-04-26:
> python opencda​​.py -tsingle_town05_cosim -v 0.9.11 并得到错误：在opencda​​/scenario_testing下找不到single_town05_cosim.py，知道吗？@DerrickXuNu

I also encountered this problem, have you solved it?

6Lackiu, 2022-04-27:
> python opencda.py -tsingle_town05_cosim -v 0.9.11 and got ERROR: single_town05_cosim.py not found under opencda/scenario_testing, any idea? @DerrickXuNu

I also encountered this problem, how did you solve it? Thanks!