ucla-mobility/OpenCDA #159: .py not found ERROR

6Lackiu, 2022-04-26:
I am trying to run opencda on a remote server with Ubuntu16.04, I had a problem with open3d before, after I solved that problem. I got the following error:
[image]
I'm sure I followed the steps in the official documentation, what should I do to fix this error?Thanks！
By the way, does opencda support running on a remote server?
Carla: 0.9.11
Driver Version: 418.43
CUDA Version: 10.1

XHwind, 2022-04-26:
Can you run `opencda.py` for other scripts in folder `opencda/scenario_testing` (e.g., `single_2lanefree_carla `). The message you get is generated in [here](https://github.com/ucla-mobility/OpenCDA/blob/4b34dc3610901484bc44c855cbc9db6a03dd079a/opencda.py#L45), which means that the opencda.py can't import the module `opencda.scenario_testing.platoon_joining_2lanefree_carla`.

6Lackiu, 2022-04-27:
> Can you run `opencda.py` for other scripts in folder `opencda/scenario_testing` (e.g., `single_2lanefree_carla `). The message you get is threw [here](https://github.com/ucla-mobility/OpenCDA/blob/4b34dc3610901484bc44c855cbc9db6a03dd079a/opencda.py#L45), which means that the opencda.py can't import the module `opencda.scenario_testing.platoon_joining_2lanefree_carla`. Please check if you have the associated script `platoon_joining_2lanefree_carla.py` in the folder `opencda/scenario_testing` and also if you have installed all the dependencies for importing the module.

Hi! I have tried all scripts but none of them work. I did follow the steps in the official documentation to install it.
[image]
[image]
I saw other people have the same problem in issues, how did they solve it?

Also, I did the same on another Ubuntu18.04 server,` python opencda.py -t platoon_joining_2lanefree_carla -v 0.9.11`, it seemed normal at first, but after a few seconds the terminal closed and even the `/OpenCDA` folder disappeared! And there is no `/OpenCDA` folder in the `Trash` folder, this confuses me, what is the reason?

DerrickXuNu, 2022-04-27:
Did you run python setup.py develop?

6Lackiu, 2022-04-27:
> Did you run python setup.py develop?

Yes I ran `python setup.py develop`.
Just now I reinstalled the environment and it still doesn't work.
This message was prompted when running `python setup.py develop`, I don't know if it had any effect.
[image]
During the whole installation process, I made only the following two changes:
1. Change open3d to open3d==0.9.0 in the environment.yml file
2. After this error: `ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type`, I added `sys.path.remove('/opt/ros /kinetic/lib/python2.7/dist-packages')` before `import cv2`
I don't know if these two points have any effect on not being able to find the .py file.

In addition, every time I run a file such as `python opencda.py -t platoon_stability_2lanefree_carla -v 0.9.11` for the first time after installing the environment, the terminal will be automatically closed, and I must restart the environment and run `opencda.py` again, and the terminal will prompt `ERROR: platoon_stability_2lanefree_carla.py not found under opencda/scenario_testing`. That's what confuses me too.

I really want to run this project. Looking forward to your reply!

XHwind, 2022-04-27:
That is wired. Could you provide more information about the error `ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type`. This error seems to indicate that you were using python2 instead of python3 and using the ros's package instead of conda's package.  
Also I think the checkpoint may be helpful for debugging. For example, you can use `pdb.set_trace()` or setting checkpoint in IDEs for the start of the `platoon_stability_2lanefree_carla.py` file. And goes line by line to see where the import fails.

6Lackiu, 2022-04-27:
> That is wired. Could you provide more information about the error `ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type`. This error seems to indicate that you were using python2 instead of python3 and using the ros's package instead of conda's package. Please double-check ros and conda compatibility. Also I think the checkpoint may be helpful for debugging. For example, you can use `pdb.set_trace()` or setting checkpoint in IDEs for the start of the `platoon_stability_2lanefree_carla.py` file. And run the code line by line to see where the import fails. Besides, you could also check if you could run (line by line) the following import statement successfully (used in the `platoon_stability_2lanefree_carla.py` file).
> 
> ```
> import sys
> import os
> import carla
> import opencda.scenario_testing.utils.sim_api as sim_api
> from opencda.core.common.misc import get_speed
> from opencda.scenario_testing.evaluations.evaluate_manager import  EvaluationManager
> from opencda.scenario_testing.utils.yaml_utils import load_yaml
> ```

Hi, unfortunately I have a new problem.
After I removed the ros related environment and files, I re-installed opencda following the documentation, but when I run the example, for example, `python opencda.py -t platoon_stability_2lanefree_carla -v 0.9.11`, then the terminal automatically quits! And the `/OpenCDA` folder disappeared with it! The deleted OpenCDA folder also doesn't appear in the recycle bin. Why does this happen? I had the same crash issue on another Ubuntu18.04 server.
Looking forward to your reply.

XHwind, 2022-04-28:
I am sorry to hear that you have this issue. We didn't observe this crash before. Could you provide more error message. I would suggest trying to set checkpoint in the `opencda.py` and `platoon_stability_2lanefree_carla.py` file to trace the line that cause the crashes and checking the associated  error message.

6Lackiu, 2022-04-28:
> 那是有线的。您能否提供有关该错误的更多信息`ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type`。此错误似乎表明您使用的是 python2 而不是 python3 并使用 ros 的包而不是 conda 的包。请仔细检查 ros 和 conda 的兼容性。 另外我认为检查点可能有助于调试。例如，您可以`pdb.set_trace()`在 IDE 中使用或设置检查点作为`platoon_stability_2lanefree_carla.py`文件的开头。并逐行运行代码，查看导入失败的地方。 此外，您还可以检查是否可以成功运行（逐行）以下导入语句（在`platoon_stability_2lanefree_carla.py`文件中使用）。
> 
> ```
> import sys
> import os
> import carla
> import opencda.scenario_testing.utils.sim_api as sim_api
> from opencda.core.common.misc import get_speed
> from opencda.scenario_testing.evaluations.evaluate_manager import  EvaluationManager
> from opencda.scenario_testing.utils.yaml_utils import load_yaml
> ```

Thanks for this reply, I seem to have found the error.
[image]
There is no yaml package in the computer. What is the function of this package? Why does it cause project errors? I'm a beginner and don't know much about this.
Also, I haven't really solved the bug, I just found it. So please don't close this issue. I need to do some other urgent work these days, and I don't have time to deal with this error. I will let you know when the problem is solved.
Finally, thank you for your patient guidance, which helped me a lot!

6Lackiu, 2022-04-30:
> I am sorry to hear that you have this issue. We didn't observe this crash before. Could you provide more error message. I would suggest trying to set checkpoint in the `opencda.py` and [...]

Hello, after I dealt with some errors, now I encountered `rpc_error`, I saw in the official carla documentation that this may be a problem with `$PYTHONPATH`, I found that my `$PYTHONPATH` is indeed a bit confusing. But after I cleared the `$PYTHONPATH` or added the path of carla0.9.11, I still encountered the error of `rpc_error`. Is there a problem with my solution?
[image]
[image]

DerrickXuNu, 2022-04-30:
Where did you install your simulator? In /opt?

6Lackiu, 2022-05-01:
> Where did you install your simulator? In /opt?

No, it is installed in my user path under `/home`
[image]
Not in `/opt`
[image]

DerrickXuNu, 2022-05-01:
Are you able to run the scripts in town06?

6Lackiu, 2022-05-02:
Sorry for taking so long to reply, the .py not found error has been resolved, it is indeed my `$PYTHONPATH` problem.
But I encountered the following problem.
[image]
I would like to ask is OpenCDA project able to run on a remotely connected server?
Ubuntu 16.04
open3d==0.9.0

6Lackiu, 2022-05-02:
> Are you able to run the scripts in town06?

Sorry for taking so long to reply, the .py not found error has been resolved, it is indeed my `$PYTHONPATH` problem.
But I encountered the following problem.
[image]
I would like to ask is OpenCDA project able to run on a remotely connected server?
Ubuntu 16.04
open3d==0.9.0

6Lackiu, 2022-05-04:
I've fixed everything and OpenCDA is working fine, thanks!

closed 2022-05-04 by DerrickXuNu