ucla-mobility/OpenCDA #166: Spawn a new CAV at a certain simulation time step

thuns001, 2022-07-15:
I was wondering if it is possible to generate a new single CAV on the on-ramp particularly for the scenario "platoon_joining_2lanefree_cosim". I tried to spawn a single cav on the on-ramp but when it reached to the merging area, about the same time as a mainline platoon (and it should perform a cut-in merge). It did not merge into the platoon. 

Please advise if OpenCDA allows us to do this. My intent is to have the simulation run longer with more CAVs. (Spawning multiple CAVs at the simulation start is possible but is limited by space of link.)

Thank you,
Thod

DerrickXuNu, 2022-07-15:
Are you trying to let 2 cavs join the same platoon or each cav joins a different platoon?

thuns001, 2022-07-15:
Each cav joins in a different platoon. Right now I only spawned one cav to see if it’s possible to do so in OpenCDA. 

Thanks

DerrickXuNu, 2022-07-15:
First, OpenCDA does support different Cavs joining different platoons, but you need to give a good start position for both so they can meet each other at the merging area. Also, I saw your YAML file, the platoon member has a too large distance gap at the beginning. Make each member in the platoon around 10 meters distance. After you pay attention to the points I mentioned and that issue still happens, you could record a video with the log output together and send to me.

DerrickXuNu, 2022-07-15:
I see. I haven't tried that before. Can you show a video with log output?

thuns001, 2022-07-16:
Hi Derrick:

Please find the attached files below for the video recording and log file.

[vdo_log_OpenCDA.zip](https://github.com/ucla-mobility/OpenCDA/files/9126291/vdo_log_OpenCDA.zip)

thuns001, 2022-07-19:
Hi Derrick,

I just wonder if you have a chance looking at the files attached above. We really hope that spawning a new cav vehicle at any time step is possible and they can cooperate with the other cavs in the network. 

thank you.

DerrickXuNu, 2022-07-19:
Hi thuns,

I have just watched the video and have several questions.
1) For the cavs that are created at the beginning, did they join the platoon successfully?
2) Can you share the code snippet that generates the new cav?

DerrickXuNu, 2022-07-20:
HI, I think respown the created vehicle may have some potential issues, as it was already in a platoon before. Have you tried created a totally new cav that is not created before?

thuns001, 2022-07-20:
I just tried spawning a new on-ramp cav from a different config yaml file. But the new CAV still did not merge into the platoon. 

            scenario_params2 = load_yaml('/home/carma3/OpenCDA/opencda/scenario_testing/config_yaml/platoon_joining_2lanefree_cosim_spawnnewcavs.yaml')
            simsec = traci.simulation.getTime()
            if simsec == 15:
                for i, cav_config in enumerate(scenario_params2['scenario']['single_cav_list']):
                    spawn_transform = map_api.spawn_helper_2lanefree('0.9.12', *cav_config['spawn_special'])
                    bp_lib = world.get_blueprint_library()
                    vehicle_bp = bp_lib.find('vehicle.tesla.model3')
                    vehicle = world.spawn_actor(vehicle_bp, spawn_transform)

                    # create vehicle manager for each cav
                    vehicle_manager = VehicleManager(vehicle, cav_config, ['platooning'], carla_map, cav_world, current_time=scenario_params2['current_time'],data_dumping=False)

                    scenario_manager.tick()

                    vehicle_manager.v2x_manager.set_platoon(None)

                    destination = carla.Location(x=cav_config['destination'][0],y=cav_config['destination'][1],z=cav_config['destination'][2])

                    vehicle_manager.update_info()
                    vehicle_manager.set_destination(vehicle_manager.vehicle.get_location(), destination, clean=True)

                    single_cav_list.append(vehicle_manager)

thuns001, 2022-07-20:
I also have another question about how can we edit the opendrive file if we want to expand the lengths of the freeway and ramp. 

Thank you,

DerrickXuNu, 2022-07-22:
> I just tried spawning a new on-ramp cav from a different config yaml file. But the new CAV still did not merge into the platoon.
>
> ```
>         scenario_params2 = [...]

i saw you append a new cav in for `i, cav_config in enumerate(scenario_params2['scenario']['single_cav_list']):` , it will actually create multiple cavs in stead of one  Also, can you make sure your created vehicle is visible in CAVWorld?

thuns001, 2022-07-26:
Hi Derrick,

My bad I actually forgot that I loaded another config yaml file, so in the file there's only one signle CAV in the list. 

Also, can you please show me how can I double-check if the newly created CAV is visible in the CAVWorld? Thank you.