ucla-mobility/OpenCDA #171: CARLA installation

ChunmianLin, 2022-08-09:
I encounter the following issue when installing CARLA with the command 'make launch' (make PythonAPI is successfully compiled): 

8 warnings and 18 errors generated.
5 warnings and 10 errors generated.
make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/admin1/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 2

Please help me!!! Many thanks.

DerrickXuNu, 2022-08-10:
Hi, I believe this issue belongs to Carla instead of OpenCDA. But you don't need to build Carla from source to use OpenCDA. You can just use the pre-build version. Have you tried that?

ChunmianLin, 2022-08-10:
Thanks. How to use prebuild  version.---- Replied Message ----FromRunsheng ***@***.***>Date08/10/2022 21:56 ***@***.***> ***@***.******@***.***>SubjectRe: [ucla-mobility/OpenCDA] CARLA installation (Issue #171)
Hi, I believe this issue belongs to Carla instead of OpenCDA. But you don't need to build Carla from source to use OpenCDA. You can just use the pre-build version. Have you tried that?

[
{
"target": "https://github.com/ucla-mobility/OpenCDA/issues/171#issuecomment-1210709687",
"url": "https://github.com/ucla-mobility/OpenCDA/issues/171#issuecomment-1210709687",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

DerrickXuNu, 2022-08-11:
You can check my personal repo to see a series of tutorials on Carla: https://github.com/DerrickXuNu/Learn-Carla

ChunmianLin, 2022-08-12:
Thanks. I follow your CARLA toturial, and when compiling 'make launch', th error occurs.
My system is Ubuntu20.04, UE4.26, and CARLA0.9.12.

closed 2022-09-20 by DerrickXuNu