ucla-mobility/OpenCDA #173: how to save the data

ChunmianLin, 2022-09-26:
Hi, author. When the simulation processing, how to save the image and point cloud data, as well as the annotation file?

DerrickXuNu, 2022-09-26:
You can check [v2xp_datadump_town06_carla.py](https://github.com/ucla-mobility/OpenCDA/blob/main/opencda/scenario_testing/v2xp_datadump_town06_carla.py) to see how to save the data.  Please notice:

1.  If you are using OpenCDA to generate and release your own dataset, then you have to get our permission to re-distribute. As clearly stated in our License, we don't allow redistribution of our codes or data generated by OpenCDA without our permission.
2. If you find our OpenCDA/Opencood/V2X-ViT is useful to your research, we'd appreciate you could star and cite our codebase.

ChunmianLin, 2022-09-27:
Many thanks. Another concern is how to select different traffic scenarios (e.g., town id) and switch various simulation settings (i.e., single-lane, platoon, etc.). There is no data generation if adopting such 'platoon_joining_2lanefree_carla' or 'single_2lanefree_carla' as testing_scenario parameter.

DerrickXuNu, 2022-09-28:
no, you have to adopt the setting in the v2xp_datadump_town06_carla.py to other scripts

closed 2022-09-28 by DerrickXuNu