ucla-mobility/OpenCDA #176: Is CARLA 0.9.9 supported?

luh-j, 2022-10-26:
Huge thanks to this great project, it looks amazing! I have a question about the supported version of Carla. I saw on the installation page, both carla 0.9.11 and 0.9.12 are supported, but due to the current projects we have to continue to use the version 0.9.9. Does your project also support carla 0.9.9? If not, would you please provide any ideas on how we could modify this great project so that it could be fitted for carla 0.9.9? Thanks!

DerrickXuNu, 2022-10-26:
It should support 0909 as well, but you may need to change the vehicle class names to 0909.

luh-j, 2022-10-27:
Thank you for much for the response, just to clarify, do you mean the blueprint of the vehicles' names or are there more places I need to check

DerrickXuNu, 2022-10-27:
Yes, this is what I mean

closed 2022-11-02 by DerrickXuNu