ucla-mobility/OpenCDA #177: OpenCDA能否导入Intereaction数据集，并将数据集中的场景进行仿真及车辆行为分析？

ShenZC25, 2022-11-28:
您好，很高兴可以了解OpenCAD。目前我只是拜读了您的文献，还没开始深入学习OpenCDA的具体操作。现在有一些问题想请问：

1.OpenCDA能否支持导入Interaction数据集，对其进行场景的还原仿真？比如复现地图，汽车驾驶轨迹，行为分析等。导入的过程中是否要对Interaction数据集中的数据类型进行转换？其他数据集呢？（InD数据集等，主要是一些汽车行为与轨迹的数据集）
2.仿真之后如果要对一些行为进行分析，或者加入一些算法进行一些研究（比如，加入LSTM进行轨迹预测，采用MPC控制动力学模型等等），能否将数据结果进行保存或者实现算法开发？

以上功能的实现，包括了OpenCDA自带的内置功能，或者我也可以自己进行算法编写（只要OpenCDA提供相应接口）。如果可以实现，我将进一步深入学习OpenCDA。

期待回复

DerrickXuNu, 2022-12-02:
Thanks for your message. Please use English to describe your issue so the whole community can understand

ShenZC25, 2022-12-05:
> Thanks for your message. Please use English to describe your issue so the whole community can understand

Hello, I'm glad to learn about OpenCDA. At present, I just read your literature, but I haven't started to study the specific operation of OpenCDA. Now I have some questions to ask:

1. Can OpenCDA support the import of Interaction data set and the restoration simulation of the scene? Such as recurring maps, car driving tracks, behavior analysis, etc. Do you want to convert the data types in the Interaction dataset during the import process? What about other data sets? (InD data sets, etc., mainly data sets of some automobile behaviors and trajectories)
2. After the simulation, if we want to analyze some behaviors, or add some algorithms to do some research (for example, adding LSTM for trajectory prediction, adopting MPC control dynamics model, etc.), can we save the data results or realize algorithm development?

The realization of the above functions includes the built-in functions of OpenCDA, or I can write the algorithm myself (as long as OpenCDA provides the corresponding interface). If it can be realized, I will further study OpenCDA.

Looking forward to your reply.

DerrickXuNu, 2022-12-14:
Here are the answers:
1) OpenCDA should be able to load it as long as it provides xodr map. But you may need to write the interface yourself to load the objects into CARLA.
2) Yes, actually, there is an evaluate_manager.py in OpenCDA repo. You can check it to see how OpenCDA evaluates the performance from different aspects.
The first one need you to write your own algorithm while second has built-in functions. But tbh, I don't think there is any tool that has built-in functions supporte  1)  in the world right now.

closed 2022-12-26 by DerrickXuNu