ucla-mobility/OpenCDA #178: Error when running with perception enabled

Shuleee, 2022-12-13:
Hello, first of all, thanks for the amazing work!
I am currently trying to run OpenCDA from the very beginning, I managed to run the very first test mentioned in the tutorial which is the Two-lane highway test with the command `python opencda.py -t single_2lanefree_carla -v 0.9.12` It works perfectly, however when I want to enable the perception with means adding the extra --apply_ml, I run with command 
`python opencda.py -t single_town06_carla  -v 0.9.11 --apply_ml`
And I got some errors which look like this:
`Connected to pydev debugger (build 223.7571.203)
OpenCDA Version: 0.1.2
Using cache found in /home/shule/.cache/torch/hub/ultralytics_yolov5_master
YOLOv5 🚀 2022-12-13 Python-3.7.10 torch-1.8.0 CUDA:0 (NVIDIA GeForce RTX 2080 Ti, 11019MiB)

Fusing layers... 
YOLOv5m summary: 290 layers, 21172173 parameters, 0 gradients, 48.9 GFLOPs
Adding AutoShape... 
Creating single CAVs.
Traceback (most recent call last):
  File "/home/shule/PycharmProjects/OpenCDA/opencda/scenario_testing/single_town06_carla.py", line 33, in run_scenario
    scenario_manager.create_vehicle_manager(application=['single'])
  File "/home/shule/PycharmProjects/OpenCDA/opencda/scenario_testing/utils/sim_api.py", line 325, in create_vehicle_manager
    data_dumping=data_dump)
  File "/home/shule/PycharmProjects/OpenCDA/opencda/core/common/vehicle_manager.py", line 109, in __init__
    map_config)
  File "/home/shule/PycharmProjects/OpenCDA/opencda/core/map/map_manager.py", line 127, in __init__
    self.generate_lane_cross_info()
  File "/home/shule/PycharmProjects/OpenCDA/opencda/core/map/map_manager.py", line 318, in generate_lane_cross_info
    tl_id = self.associate_lane_tl(mid_lane)
  File "/home/shule/PycharmProjects/OpenCDA/opencda/core/map/map_manager.py", line 268, in associate_lane_tl
    trigger_path = Path(trigger_poly.boundary)
  File "/home/shule/anaconda3/envs/opencda/lib/python3.7/site-packages/matplotlib/path.py", line 127, in __init__
    vertices = _to_unmasked_float_array(vertices)
  File "/home/shule/anaconda3/envs/opencda/lib/python3.7/site-packages/matplotlib/cbook/__init__.py", line 1317, in _to_unmasked_float_array
    return np.asarray(x, float)
TypeError: float() argument must be a string or a number, not 'LineString'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/shule/PycharmProjects/OpenCDA/opencda.py", line 56, in <module>
    main()
  File "/home/shule/PycharmProjects/OpenCDA/opencda.py", line 51, in main
    scenario_runner(opt, config_yaml)
  File "/home/shule/PycharmProjects/OpenCDA/opencda/scenario_testing/single_town06_carla.py", line 64, in run_scenario
    eval_manager.evaluate()
UnboundLocalError: local variable 'eval_manager' referenced before assignment
python-BaseException
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 1109 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 1110 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 1114 (sensor.camera.rgb) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 1113 (sensor.camera.rgb) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 1112 (sensor.camera.rgb) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 1111 (sensor.camera.rgb) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 1115 (sensor.lidar.ray_cast) 

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)`
I am running under Ubuntu 20.04 and my pytorch version is 1.8. Plus I am amateur for machine learning so I just follow the steps to install yolov5 and have done nothing else.
Looking forward for your answers, thanks in advance.

Best,
Shule

YuanYunshuang, 2022-12-14:
If you are using shapely 2.0, downgrade it  to a lower version, i.e. 1.8.1

Shuleee, 2022-12-14:
Thx Yunshuang, I downgraded it to 1.8.4 and is working now. I suggest changing this also in the requirement.txt so that no one will have this problem again.

closed 2022-12-14 by Shuleee