ucla-mobility/OpenCDA #180: Where to find the coordinates for a spawn location

JeanPhilippeMonteuuis, 2023-01-25:
Hello,

My goal is to spawn the ego-vehicle at a specific location in Town07 (e.g., singleTown07_carla).
In the picture, the red dot is the `spawn position`.
The blue dot is the `destination`.
[image]

I do not understand where/how to retrieve the coordinates used in `spawn_position` and `destination` (see lines below)

scenario:
  single_cav_list: # this is for merging vehicle or single cav without v2x
    - <<: *vehicle_base
      spawn_position: [600, 51, 50, 0, 0, 0]
      destination: [600, 145.51, 50]

I have tried to used the x, y , z coordinates provided in the Unreal Engine Editor but it seems off.
Can you provide me some guidance or an example?

Thank you!

DerrickXuNu, 2023-02-20:
Thanks for using OpenCDA! I am a little confused about your question. What do you mean "it seems off"?

JeanPhilippeMonteuuis, 2023-02-21:
Hey @DerrickXuNu ,
Thank you for your reply.

I would like to understand: How/Where do you pick x,y,z for `spawn_position` (red dot in the figure) and `destination` (blue dot in the figure) to fit my case (figure above)?

About "It seems off", it means to retrieve x,y,z coordinates:
- I open the Unreal Editor
- I load the Town07 (in Unreal Editor)
- I check the x,y,z coordinates of an asset near my starting spawn point (red dot) and my ending spawn point (blue dot).
- Back to OpenCDA, I enter the x,y,z  coordinates in`spawn_position` and `destination`
- I launch my custom OpenCDA scenario (e.g., `python opencda.py -t single_town07_carla  -v 0.9.12 --apply_ml`
- The carla server is loaded
- However, I do not see the car.

DerrickXuNu, 2023-02-28:
Thanks for clarifying your question.

 How/Where do you pick x,y,z for spawn_position (red dot in the figure) and destination (blue dot in the figure) to fit my case (figure above)? --> I used UE4Editor (build froun source version) to find the available spawn points.

It seems off -> there are several things you want to try:
1) Make sure you set the spectator to the vehicle you spawn, so that the CarlaUE4 will focus on it automatically
2) Tune the Z value. It maybe your Z is too low so it can not spawn. 
3) Check whether there is some error message when you start your script

closed 2023-03-08 by DerrickXuNu