ucla-mobility/OpenCDA #202: Controller selection in OpenCDA

hlwei, 2023-04-24:
How to specify the controller in OpenCDA? I know there is a PID controller in the source code. But I cannot find where you define the controller type. Could you please provide some comments and help on this? Thx.

wangxso, 2023-08-03:
in the `default.yaml` you can specify the controller like this

```yaml
 controller:
    type: pid_controller # this has to be exactly the same name as the controller py file
    args:
      lat:
        k_p: 0.75
        k_d: 0.02
        k_i: 0.4
      lon:
        k_p: 0.37
        k_d: 0.024
        k_i: 0.032
      dynamic: false # whether use dynamic pid setting
      dt: ${world.fixed_delta_seconds} # this should be equal to your simulation time-step
      max_brake: 1.0
      max_throttle: 1.0
      max_steering: 0.3
```

and you need write some controller algorithm in `opencda/core/actuation/xxx_controller.py`
I don't know if it's right or not, I hope it can help you.

wangxso, 2023-08-03:
> in the `default.yaml` you can specify the controller like this
>
> ```yaml
>  controller:
>     type: pid_controller # this has to be exactly the same name as the controller py file
>     args: [...]

and you also can read the doc with [customization](https://opencda-documentation.readthedocs.io/en/latest/md_files/customization.html)

closed 2023-08-05 by DerrickXuNu