ucla-mobility/OpenCDA #209: Insallation Problems (local installation and docker)

mcitir, 2023-06-08:
Hello, 
I want to use OpenCDA and make some tests for V2X scenarios. However, I have encountered installation problems even I complete all steps correctly. 

### Errors for Local installation
1.  I have an conda environment and install all dependencies correctly without any error. 
2. Also, I installed carla prebuild version, and extracted additional maps.

If I run `single_2lanefree_carla`, I got an error as below:

[image]

I also tested with another scenario `single_town06_carla` but I got extra error with the same error. However, the maps from [link](https://github.com/carla-simulator/carla/blob/master/Docs/download.md) are already downloaded and extracted to Carla with `./ImportAssets.sh` command. I can see the town06 at the folder as below:

[image]

However, I still get the following error and the `eval_manager` error is still present:

[image]

### Errors for Docker installation
First small error, during the build of docker image, the name was selected as `opencda_container`, but then in the running command, it is called as `opencda_docker`, so it produces an error

[image]

I had set `OPENCDA_FULL_INSTALL` to `true`, I did not run `setup.sh`
Then I realized the python version of carla dist is not correct:

[image]

Naturally, `carla module` could not found:

[image]

Then, I run `setup.sh` to install carla

[image]
 
Then, I got an version error about numpy and I upgraded,

[image]

After that, when I run the scenario, I got exact same error with local installation:

DerrickXuNu, 2023-06-08:
It seems that you didn't run Carla. Did you run Carla first and run OpenCDA?

mcitir, 2023-06-08:
> It seems that you didn't run Carla. Did you run Carla first and run OpenCDA?
Yes, the carla is already running.
[image]

DerrickXuNu, 2023-06-08:
Did you install your Carla under /opt path?

Soe518, 2023-08-11:
[image]

> Did you install your Carla under /opt path?

I have the same problem and don't know how to solve it as well ...