ucla-mobility/OpenCDA #210: float() argument must be a string or a number, not 'LineString' in single_intersection_town6_carla file

SExpert12, 2023-06-10:
I am working in Carla simulator. I required specific scenario for my simulation. So I have tried to run file named single_intersection_town06_carla and it it showing me error like "float() argument must be a string or a number, not 'LineString'". I traced the error and I found that in the file named core/map/map_manager.py the input is actually linestring. But I am not able to understand that how to solve this as shapely polygon and boundary function of polygon are completely new to me.  How to resolve this error?
Here  I have written the same function of the file map_manager.py:
def associate_lane_tl(self, mid_lane):
        """
        Given the waypoints for a certain lane, find the traffic light that
        influence it.

        Parameters
        ----------
        mid_lane : np.ndarray
            The middle line of the lane.
        Returns
        -------
        associate_tl_id : str
            The associated traffic light id.
        """
        associate_tl_id = ''

        for tl_id, tl_content in self.traffic_light_info.items():
            
            trigger_poly = tl_content['corners']
            print("\n\n corners",tl_content['corners'])
            print("\n\n trigger_poly",trigger_poly) # printing values
            print(type(trigger_poly))
            # use Path to do fast computation

            trigger_path = Path(float(trigger_poly.boundary))

            print("trigger_path",type(trigger_poly.boundary))

            # check if any point in the middle line inside the trigger area
            check_array = trigger_path.contains_points(mid_lane[:, :2])

            if check_array.any():
                associate_tl_id = tl_id
        return associate_tl_id
[image]

SExpert12, 2023-06-15:
Name: shapely
Version: 2.0.1

DerrickXuNu, 2023-06-15:
Yes, this is the reason. Please stick to the shapely version appeared in the requirement.txt

SExpert12, 2023-06-15:
Okay. But then I am getting problem with other packages.

closed 2023-06-23 by DerrickXuNu