ucla-mobility/OpenCDA #214: installation document should be updated

2019211753, 2023-08-29:
Numpy>=1.22 required by yolo5 don't support Python3.7 anymore. But the suggested Python version in your document is still 3.7.

https://numpy.org/devdocs/release/1.22.0-notes.html#:~:text=The%20Python%20versions%20supported%20in%20this%20release%20are%203.8%2D3.10%2C%20Python%203.7%20has%20been%20dropped

2019211753, 2023-08-29:
'python setup.py develop' throws 'Searching for pandas
Reading https://pypi.org/simple/pandas/
Downloading https://files.pythonhosted.org/packages/a9/ec/e531733ff7f955e58ac45089cf72d085e587a731778b9dac1b82878815e9/pandas-2.1.0rc0.tar.gz#sha256=e6967a82ee26997ab5ef52907064ca21eed5e51089dc2131a252a26d67d88b36
Best match: pandas 2.1.0rc0
Processing pandas-2.1.0rc0.tar.gz
error: Couldn't find a setup script in /tmp/easy_install-5gxr_ca2/pandas-2.1.0rc0.tar.gz' after switching to Python3.8

ayton-zhang, 2023-11-21:
I encountered the same numpy issue when I tried to install yolo5.

2019211753, 2023-11-25:
> I encountered the same numpy issue when I tried to install yolo5.

just switch to earlier version, such as tag v6.2

Sumail2020, 2024-03-28:
> Numpy>=1.22 required by yolo5 don't support Python3.7 anymore. But the suggested Python version in your document is still 3.7. [...]

You can directly use "pip install yolov5" to fix the problem