ucla-mobility/OpenCDA #215: Is the branch of cooperative-lidar-communicate really realize lidar data transmit between CAVs?

kanglang123, 2023-09-13:
I just find it realize that ego can find nearby car by v2x manager when perception activate mode is set to false.

this part of code is at opencda/core/sensing/perception/perception_manager.py line:486-490 of opencda/core/sensing/perception/perception_manager.py

` if not self.activate: 
  self.search_nearby_cav()
            objects = self.deactivate_mode(objects)    # maybe 当不检测的时候通过v2x来查到周围的车的位置来作为障碍物的位置
        else:
            objects = self.activate_mode(objects)
`

could you explain which part of code implements the function of **cooperative lidar communicate**?
I spent a long time not finding it.

wangxso, 2023-09-14:
I found the method `search_nearby_cav()`  in the branch `cooperative-lidar-communicate`
```python
def search_nearby_cav(self):
        if self.v2x_manager is not None:
            print(f"nearby cav nearby {self.v2x_manager.cav_nearby}")
            for  _, vm in self.v2x_manager.cav_nearby.items():
                lidar = vm.v2x_manager.get_ego_lidar()
                print(f"lidar {lidar}")
        pass
```
When I contemplate it, I believe this is not a complete function.

kanglang123, 2023-09-14:
> I found the method `search_nearby_cav()` in the branch `cooperative-lidar-communicate`
>
> ```python
> def search_nearby_cav(self):
>         if self.v2x_manager is not None:
>             print(f"nearby cav [...]

In v2x_manager.py, I find that author add the functions named  "get_ego_lidar" and "get_ego_rgb_image", I think we can use this two function to get perception information from other CAVs, by this way, it realize the information sharing. 
It's just that I have not find author use the two function to do some meaningful things.

closed 2023-09-16 by kanglang123