ucla-mobility/OpenCDA #222: Request for OPV2V data collection code

flow233-ai, 2023-12-29:
Hi! Recently I am doing some research on vehicular networks. I wonder whether there are some ways to generate a dataset like OPV2V.
I found the statement in OpenCOOD tutorial as follows:

'To help users collect customized data in CARLA (e.g., different sensor configurations and modalities) with similar data format and structure, we have released our OPV2V data collection code in the feature/data_collection branch of [OpenCDA](https://github.com/ucla-mobility/OpenCDA/tree/feature/data_collection).'

But I didn't find the "feature/data_collection" branch of OpenCDA. I wonder which part of the OpenCDA repository is now the part about data collection for OPV2V.

Thank you in advance for any help you might be able to provide.

Oshwiciqwq, 2024-01-08:
You can see this issue https://github.com/ucla-mobility/OpenCDA/issues/219

le-liang, 2024-01-09:
> You can see this issue #219

OK. Thank you so much.

closed 2024-07-05 by DerrickXuNu