ucla-mobility/OpenCDA #224: UnboundLocalError local variable 'eval_manager' referenced before assignment

Wanchor0729, 2024-03-21:
跪求大佬们解决！！！
发生异常: UnboundLocalError
local variable 'eval_manager' referenced before assignment

jason696, 2024-05-14:
Same problem. Any update or seeking help from dalao

DerrickXuNu, 2024-07-05:
Hi, can you put the full error message here?

kingDanonymous, 2024-07-08:
> Hi, can you put the full error message here?

Hi,
I have found this error message:

YOLOv5 🚀 2024-5-28 Python-3.7.10 torch-1.13.1+cu117 CUDA:0 (NVIDIA GeForce RTX 3070, 7974MiB)

Fusing layers... 
YOLOv5m summary: 290 layers, 21172173 parameters, 0 gradients
Adding AutoShape... 
 Retrying in 1 seconds
/home/carma/mamun/OpenCDA/opencda/co_simulation/sumo_integration/sumo_simulation.py:297: FutureWarning: This search incorrectly ignores the root element, and will be fixed in a future version.  If you rely on the current behaviour, change it to './/net-file'
  tag = tree.find('//net-file')
Traceback (most recent call last):
  File "/home/carma/mamun/OpenCDA/opencda/scenario_testing/single_town06_cosim.py", line 36, in run_scenario
    sumo_file_parent_path=sumo_cfg)
  File "/home/carma/mamun/OpenCDA/opencda/scenario_testing/utils/cosim_api.py", line 105, in __init__
    sumo_client_order)
  File "/home/carma/mamun/OpenCDA/opencda/co_simulation/sumo_integration/sumo_simulation.py", line 335, in __init__
    self.net = _get_sumo_net(cfg_file)
  File "/home/carma/mamun/OpenCDA/opencda/co_simulation/sumo_integration/sumo_simulation.py", line 304, in _get_sumo_net
    sumo_net = traci.sumolib.net.readNet(net_file)
AttributeError: module 'traci' has no attribute 'sumolib'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "opencda.py", line 75, in <module>
    main()
  File "opencda.py", line 70, in main
    scenario_runner(opt, scene_dict)
  File "/home/carma/mamun/OpenCDA/opencda/scenario_testing/single_town06_cosim.py", line 65, in run_scenario
    eval_manager.evaluate()
UnboundLocalError: local variable 'eval_manager' referenced before assignment

2410967417, 2024-12-12:
(opencda) E:\OpenCDA\OpenCDA-main>python opencda.py -t single_town06_cosim  -v 0.9.12 --apply_ml
OpenCDA Version: 0.1.3
Using cache found in C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master
Traceback (most recent call last):
  File "E:\OpenCDA\OpenCDA-main\opencda\scenario_testing\single_town06_cosim.py", line 22, in run_scenario
    cav_world = CavWorld(opt.apply_ml)
  File "E:\OpenCDA\OpenCDA-main\opencda\core\common\cav_world.py", line 53, in __init__
    self.ml_manager = ml_manager()
  File "E:\OpenCDA\OpenCDA-main\opencda\customize\ml_libs\ml_manager.py", line 29, in __init__
    self.object_detector = torch.hub.load('ultralytics/yolov5', 'yolov5m')
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\torch\hub.py", line 542, in load
    model = _load_local(repo_or_dir, model, *args, **kwargs)
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\torch\hub.py", line 572, in _load_local
    model = entry(*args, **kwargs)
  File "C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master\hubconf.py", line 248, in yolov5m
    return _create("yolov5m", pretrained, channels, classes, autoshape, _verbose, device)
  File "C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master\hubconf.py", line 56, in _create
    from models.yolo import ClassificationModel, DetectionModel, SegmentationModel
  File "C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master\models\yolo.py", line 55, in <module>
    from utils.plots import feature_visualization
  File "C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master\utils\plots.py", line 15, in <module>
    import seaborn as sn
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\seaborn-0.13.2-py3.7.egg\seaborn\__init__.py", line 2, in <module>
    from .rcmod import *  # noqa: F401,F403
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\seaborn-0.13.2-py3.7.egg\seaborn\rcmod.py", line 5, in <module>
    from . import palettes
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\seaborn-0.13.2-py3.7.egg\seaborn\palettes.py", line 9, in <module>
    from .utils import desaturate, get_color_cycle
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\seaborn-0.13.2-py3.7.egg\seaborn\utils.py", line 441
    if (new_labels := kwargs.pop("labels", None)) is not None:
                   ^
SyntaxError: invalid syntax

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "opencda.py", line 75, in <module>
    main()
  File "opencda.py", line 70, in main
    scenario_runner(opt, scene_dict)
  File "E:\OpenCDA\OpenCDA-main\opencda\scenario_testing\single_town06_cosim.py", line 69, in run_scenario
    eval_manager.evaluate()
UnboundLocalError: local variable 'eval_manager' referenced before assignment
我也有这个问题，大佬能来解决一下吗

2410967417, 2024-12-12:
(opencda) E:\OpenCDA\OpenCDA-main>python opencda.py -t single_town06_cosim  -v 0.9.12 --apply_ml
OpenCDA Version: 0.1.3
Using cache found in C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master
Traceback (most recent call last):
  File "E:\OpenCDA\OpenCDA-main\opencda\scenario_testing\single_town06_cosim.py", line 22, in run_scenario
    cav_world = CavWorld(opt.apply_ml)
  File "E:\OpenCDA\OpenCDA-main\opencda\core\common\cav_world.py", line 53, in __init__
    self.ml_manager = ml_manager()
  File "E:\OpenCDA\OpenCDA-main\opencda\customize\ml_libs\ml_manager.py", line 29, in __init__
    self.object_detector = torch.hub.load('ultralytics/yolov5', 'yolov5m')
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\torch\hub.py", line 542, in load
    model = _load_local(repo_or_dir, model, *args, **kwargs)
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\torch\hub.py", line 572, in _load_local
    model = entry(*args, **kwargs)
  File "C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master\hubconf.py", line 248, in yolov5m
    return _create("yolov5m", pretrained, channels, classes, autoshape, _verbose, device)
  File "C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master\hubconf.py", line 56, in _create
    from models.yolo import ClassificationModel, DetectionModel, SegmentationModel
  File "C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master\models\yolo.py", line 55, in <module>
    from utils.plots import feature_visualization
  File "C:\Users\Administrator/.cache\torch\hub\ultralytics_yolov5_master\utils\plots.py", line 15, in <module>
    import seaborn as sn
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\seaborn-0.13.2-py3.7.egg\seaborn\__init__.py", line 2, in <module>
    from .rcmod import *  # noqa: F401,F403
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\seaborn-0.13.2-py3.7.egg\seaborn\rcmod.py", line 5, in <module>
    from . import palettes
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\seaborn-0.13.2-py3.7.egg\seaborn\palettes.py", line 9, in <module>
    from .utils import desaturate, get_color_cycle
  File "C:\ProgramData\miniconda3\envs\opencda\lib\site-packages\seaborn-0.13.2-py3.7.egg\seaborn\utils.py", line 441
    if (new_labels := kwargs.pop("labels", None)) is not None:
                   ^
SyntaxError: invalid syntax

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "opencda.py", line 75, in <module>
    main()
  File "opencda.py", line 70, in main
    scenario_runner(opt, scene_dict)
  File "E:\OpenCDA\OpenCDA-main\opencda\scenario_testing\single_town06_cosim.py", line 69, in run_scenario
    eval_manager.evaluate()
UnboundLocalError: local variable 'eval_manager' referenced before assignment
我也有这个问题，大佬能告诉我一下解决方法吗