ucla-mobility/OpenCDA #235: Conflict between opencda and yolov5 for numpy version

cyKkk0, 2025-02-25:
In requirements.txt, opencda need **numpy==1.20.0** and **python 3.7**, while in ducoment, when download the requirements for yolov5, it needs **numpy>=1.23.5** which is not supported by python 3.7, how to solve this conflict?

payprays, 2025-03-08:
the same is https://github.com/ucla-mobility/OpenCDA/issues/214

cyKkk0, 2025-03-09:
> the same is [#214](https://github.com/ucla-mobility/OpenCDA/issues/214)

oh no, I didn't find it at all, thank you. And alongside this way, I just found another way to solve this problem. 

I just clone the yolov5 and checkout to the tag v6.2 and the requirements are met, just `pip install -r requirements.txt`

When running the command `python opencda.py -t single_town06_carla  -v 0.9.12 --apply_ml`, it will run the following command `Downloading: "https://github.com/ultralytics/yolov5/archive/master.zip" to /home/bupt/.cache/torch/hub/master.zip`, then you can find **ultralytics_yolov5_master** folder in `~/.cache/torch/hub/`, then replace the file in folder with yolov5 with tag v6.2.

closed 2025-03-09 by cyKkk0