ucla-mobility/OpenCDA #236: Running the latest commit code with sample args occurs error

payprays, 2025-03-08:
## Environment
- OS: Ubuntu 22.04
- Python: 3.7.12 (Anaconda environment)
- matplotlib: 3.4.2
- OpenCDA version: 0.1.3
- CARLA version: 0.9.12

## Problem Description
When running the platoon joining scenario with co-simulation, I'm getting a dimension mismatch error in the evaluation manager when trying to plot data:

ValueError: x and y must have same first dimension, but have shapes (1124,) and (1126,)

The error occurs in the `plot_3d` method of the `EvaluationManager` class when trying to plot accelerometer data. The time series (x-axis) has 1124 data points while the accelerometer data (y-axis) has 1126 points.

## Steps to Reproduce
1. Clone the repository
2. Run the platoon joining scenario:
   ```
   python opencda.py -t platoon_joining_2lanefree_cosim -v carla_version --apply_ml --record -v 0.9.12
   ```

## Error Log
The error occurs during the evaluation phase after the simulation completes successfully:
```
***********Planning Evaluation Module***********
Planned distance: 1343.6927490234375
Real distance: 1341.3471999624744
...
ValueError: x and y must have same first dimension, but have shapes (1124,) and (1126,)
```

Additionally, I notice the warning:
```
UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
```

here is the complete console:
```
est@4090-pc ~/V/O/OpenCDA (main)>  cd /home/test/V2X/OpenCDA/OpenCDA ; /usr/bin/env /home/test/anaconda3/envs/opencda
/bin/python /home/test/.vscode/extensions/ms-python.debugpy-2024.2.0-linux-x64/bundled/libs/debugpy/adapter/../../debu
gpy/launcher 59233 -- /home/test/V2X/OpenCDA/OpenCDA/opencda.py -t platoon_joining_2lanefree_cosim -v carla_version --
apply_ml --record -v 0.9.12 
OpenCDA Version: 0.1.3
Using cache found in /home/test/.cache/torch/hub/ultralytics_yolov5_master
YOLOv5 🚀 v6.2-0-gd3ea0df Python-3.7.12 torch-1.8.0 CUDA:0 (NVIDIA GeForce RTX 4090, 24109MiB)

Fusing layers... 
YOLOv5m summary: 290 layers, 21172173 parameters, 0 gradients, 48.9 GFLOPs
Adding AutoShape... 
load opendrive map '2lane_freeway_simplified.xodr'.
Starting new sumo server...
Remember to press the play button to start the simulation
 Retrying in 1 seconds
/home/test/V2X/OpenCDA/OpenCDA/opencda/co_simulation/sumo_integration/sumo_simulation.py:297: FutureWarning: This search incorrectly ignores the root element, and will be fixed in a future version.  If you rely on the current behaviour, change it to './/net-file'
  tag = tree.find('//net-file')
Creating platoons/
Using cache found in /home/test/.cache/torch/hub/ultralytics_yolov5_master
YOLOv5 🚀 v6.2-0-gd3ea0df Python-3.7.12 torch-1.8.0 CUDA:0 (NVIDIA GeForce RTX 4090, 24109MiB)

Fusing layers... 
YOLOv5m summary: 290 layers, 21172173 parameters, 0 gradients, 48.9 GFLOPs
Adding AutoShape... 
Creating single CAVs.
sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.bmw.grandtourer
sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.audi.tt
/home/test/V2X/OpenCDA/OpenCDA/opencda/core/application/platooning/platoon_behavior_agent.py:303: RuntimeWarning: divide by zero encountered in double_scalars
  time_gap = distance / delta_v
sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.bmw.grandtourer
sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.audi.a2
sumo vtype DEFAULT_VEHTYPE not found in carla. The following blueprint will be used: vehicle.nissan.patrol
[... 339 lines cut ...]
    raise ValueError(f"x and y must have same first dimension, but "
ValueError: x and y must have same first dimension, but have shapes (1124,) and (1126,)
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 460 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 461 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 462 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 463 (sensor.other.collision) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 480 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 481 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 482 (sensor.camera.rgb) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 483 (sensor.lidar.ray_cast) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 484 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 485 (sensor.other.collision) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 475 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 476 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 477 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 478 (sensor.other.collision) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 470 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 471 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 472 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 473 (sensor.other.collision) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 465 (sensor.other.gnss) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 466 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 467 (sensor.other.imu) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 468 (sensor.other.collision) 
```

payprays, 2025-03-08:
About the warnning
```
UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
```
I tried to add the code in the beginning of evaluate_manager.py with the same result.
```
import matplotlib
matplotlib.use('Qt5Agg')
```
I download it to file manually.

[image]

payprays, 2025-05-22:
It may be caused by improper destroying of sensors.

see the pr link to solve it.
https://github.com/ucla-mobility/OpenCDA/pull/238#issue-2937933808

WinouyFeng, 2025-05-22:
谢谢大佬，我看你是西电的，方便进一步请教一下吗，我试了你发的那个fix版本还是出现一样的问题，折磨了我好多天了，实在没办法解决，跪求指点，谢谢谢谢。

payprays, 2025-05-23:
我fork了这个，在我的fuzz分支下改了些东西，你可以看看我的仓库，大概一个月前写的了，有些东西没啥印象了。

WinouyFeng, 2025-05-23:
好的好的，感谢感谢！

Jeremath, 2025-08-05:
Thank you for solving the problem I encountered when using OpenCDA simulation test.